nikivanov/watney #11: Speech function?

farminguin, 2019-03-10:
Would it be possible to add eSpeak Text-To-Speech? Like a LM4871 audio amp ? Textbox on the web interface? Would be a great way to make it for of a telepresence bot. ..

svasiliou, 2019-03-24:
There is a 2way-audio branch but need explanation and wiring management.

nikivanov, 2019-04-08:
To get 2 way audio working, I had to switch to Arch Linux from Raspbian. The 2wayaudio branch isn't functional yet, so I'd avoid using it for now. Once it's ready, I'll release an SD card image along with BOM and 3d printed parts. Overall, the cost of adding 2 way audio is about $6 if you order from China (plus printed parts).

nikivanov, 2019-10-15:
2 way audio has been implemented

closed 2019-10-15 by nikivanov