nikivanov/watney #12: Smartphone/Tablet control

svasiliou, 2019-03-13:
Project is awesome.
I believe that adding a few buttons (up/down/left/right/tilt) on lower  button row, would add functionality with tablet or smartphone. Is that possible or need complete redesign of code?

nikivanov, 2019-04-08:
Thanks! I'm currently working on adding 2 way audio so you can use Watney as a presence bot. After that I want to add wireless charging so it could be used indefinitely without needing to plug it in. For the latter part, I've been trying to figure out how to do voltage monitoring with Raspberry Pi - I'll look into the ADC you listed, thanks!

Adding a D-pad control for mobile interface has been on my list for a while, but I haven't gotten to it yet. Stay tuned!

svasiliou, 2019-04-08:
There are two interesting virtual joystick scripts
https://github.com/jeromeetienne/virtualjoystick.js (a little complex but more powerful)
https://github.com/cptx032/virtual-joystick (a simple one and I believe more suitable for WATNEY)

nikivanov, 2023-03-16:
@baheamri what would you like to know?