nikivanov/watney #14: No video

IgnacioHR, 2019-09-22:
Hello,

First of all, thank you for the material posted here. I'm having fun and learning a lot trying to build one Watney4

At this time, I've printed all parts and I've also modified some of them in order to adapt to the servo (GH-S43A) which is the only one I found that matches the required dimensions.

I'm still waiting for the motors for the wheels so I decided to test the other pieces separately from the entire robot.

The main problem I'm having is with the video. I've spent more than 4 days reading, compiling, changing things testing etc... with very little success.

1) I started using the software published on the "master" branch, but it looks like the video protocol is a security hole and not supported by browsers. So I realised I'd to go for the second branch (2wayaudio)

2) I moved to the new 2wayaudio branch. Downloaded janus from source (and several other libraries) tested if the video camera works and the scripts... and I'm stuck with a blank screen on my browser.

I'm still using a Raspberry pi zero W. It has no sound card so I can't test the sound. It might be good if sound is an option in the software so we could test things step by step.

Regarding the video. I've been able to save a sample video file in h264 format, upload the file to my MAC and then reproduce it on VLC so the hardware works and the first part of the video.sh script to capture the images from the capera works. OK.

I cannot test the second part of the pipe in the script. I've tried to use VLC on the Raspberry pi to implement the server side and test if the network is not a problem and I can see the video (with bad FPS...) but from time to time I can see images coming to VLC Client on the MAC.

But, when I start JANUS to test all pieces together, I'm stuck in a black screen with no images. The janus.log file says:

```[Sun Sep 22 23:49:37 2019] [WARN] [1962729996216110] No stream, queueing this trickle as it got here before the SDP...
[Sun Sep 22 23:49:37 2019] [1962729996216110] Creating ICE agent (ICE Full mode, controlled)
[Sun Sep 22 23:49:38 2019] [ERR] [sdp-utils.c:janus_sdp_get_codec_rtpmap:763] Unsupported codec 'none'
[Sun Sep 22 23:50:17 2019] [ERR] [dtls.c:janus_dtls_retry:1029] [2445920852038508] DTLS taking too much time for component 1 in stream 1...
[Sun Sep 22 23:50:17 2019] [janus.plugin.streaming-0xb2f08938] No WebRTC media anymore
[Sun Sep 22 23:50:17 2019] [2445920852038508] WebRTC resources freed; 0xb2f07108 0xb2f02ca0
[Sun Sep 22 23:50:17 2019] [ERR] [dtls.c:janus_dtls_retry:1029] [1962729996216110] DTLS taking too much time for component 1 in stream 1...
[Sun Sep 22 23:50:17 2019] [janus.plugin.videoroom-0xb2f051e8] No WebRTC media anymore; 0xb2f09e78 0xb2f06520
[Sun Sep 22 23:50:17 2019] [1962729996216110] WebRTC resources freed; 0xb2f09e78 0xb2f02ca0
[Sun Sep 22 23:50:17 2019] [WARN] [1962729996216110] No stream, queueing this trickle as it got here before the SDP...
[```

and I don't know how to move forward...

Cheers
Ignacio

nikivanov, 2019-10-15:
Sorry I'm getting back to you so late.

I released a new version of Watney yesterday - this one definitely works, at least in Chrome. However, you need RPi 3A+, since Zero W doesn't have enough oomph to do both audio and video. Also I switched to a more popular SG90 servo.

Let me know if you're still trying to get the old Watney to work and I could help you troubleshoot it.

IgnacioHR, 2019-10-16:
Thank you for your response!

I do have some RPi 3A and 4 for testing and I will checkout the new version of the code and see where I can go.

Cheers
Ignacio

closed 2020-04-23 by nikivanov