nikivanov/watney #16: Feature Request: LED Headlights

rtaylo, 2019-12-09:
I'm not sure where to put a feature request in, but I'd be happy to help work on this.
I think some bright LEDs in the front would be perfect for navigating dark spaces.

nikivanov, 2019-12-23:
You'd need to figure where to get power and how to toggle the LEDs.

For power, you have a spare 3V output on the power board and a 5V pin on the raspberry pi. You'd have to measure the current on both to find where you have more headroom. 5V would be easier, but it might already be pretty taxed. 3V is safer but more complicated when it comes to toggling.

For toggling LEDs on and off you could use a relay/mosfet, or use an individually-controlled LED strip, like WS2801 that would allow toggling natively. I think WS2801 is the easiest approach, as long as there's some headroom. The power board max output at 5V is 3A, but they recommend not going above 2.2A.

nikivanov, 2020-04-23:
I'm planning on adding headlights based on SK9822 LED strip. With 50mA per LED and 30/m strip, I think I can have 4 LEDs wrap around the front of Watney and provide enough illumination. Stay tuned! Closing this request for now.

closed 2020-04-23 by nikivanov