nikivanov/watney #23: Post your Watney builds here!

nikivanov, 2021-11-17:
Hi! Please post pictures of your Watney builds here.

rtaylo, 2022-02-01:
Video of Watney 2.0 with Flashlight attachment. 

https://user-images.githubusercontent.com/21061419/152043389-0781231a-5ca4-4187-8e41-0215fb3ea45e.mp4

Syndesi, 2022-02-14:
Not a Watney robot, but similar in size and maybe an inspiration for a mod?
https://hackaday.com/2022/02/10/a-simple-3d-printed-rover-design
[image]
/

sarreck, 2022-03-14:
Just finished my build and its working really well. 

Green is PETG, Yellow is ABS, black TPU for the tires. 

Thanks for the great design
[image]
[image]

[image]
[image]
[image]

rtaylo, 2022-03-14:
[image]

Here's my garage. Open slot for new Watney 3.0

rtaylo, 2022-04-10:
https://user-images.githubusercontent.com/21061419/162643409-6dd32fcf-e642-4c96-9b4e-c07bddfb0bd7.mp4

rtaylo, 2022-04-10:
[image]

Spangen, 2022-12-08:
https://user-images.githubusercontent.com/4445078/206438544-97e8863e-fb91-4e12-a747-3627ae77e471.mp4

4 x hoverboard motors, 2 x reflashed hoverboard controller boards as motor drivers. 4 x 350W off torque from standstill and 4wd ;) I Put a chair on this thing, and sat on it. When I hit the trottle, I was thrown off ;)

Tried your new image to get to know webrtc and h264 streaming, and se how you solved audio. Is going to be great to ditch mjpeg-streamer on my 4G rover, it eats bandwidth ;)

Would be great with support for comute module 4, so we could use 2 cams. Reverse cam would be a great addon. Have that on another robot build I run on a modified Zerobot software setup. 

Greatings from Norway

73 de LA5XOA

rtaylo, 2022-12-08:
Ha this is awesome

EGimagination, 2023-04-28:
This has been a Fantastic project, Thank you for sharing. Here is my Watney build, currently living within our local library that I volunteer at helping kids get into 3D printing and STEM learning.
[image]
[image]
[image]
[image]

fredmetal, 2023-12-08:
Nice Project, could the motor controllers be changed for L298N?

nikivanov, 2023-12-08:
> Nice Project, could the motor controllers be changed for L298N?

@fredmetal  I used L298N in previous versions of Watney, but it causes a fairly significant voltage drop. With 5V on the input, you lose quite a lot of torque due to the V drop, so I swapped it out for DRV8833. You can certainly go back to using L298N, but im not sure if there's enough space in the current design to accomodate for it.

fredmetal, 2023-12-08:
@nikivanov, I understand, I will continue with the DRV8833, thanks for sharing your project, crack!

nikivanov, 2023-12-08:
@fredmetal  Sure thing. Please open a separate issue if you have any questions or problems with assembly. Just a heads up - I'll be deleting this conversation from this thread in a bit - I'd like to keep it for build pictures only.

eerison, 2023-12-10:
[image]

Thank you for this nice project ♥️

nikivanov, 2023-12-11:
@eerison nice build! What kind of camera did you use? It looks like it doesn't quite fit into the housing.

eerison, 2023-12-11:
> @eerison nice build! What kind of camera did you use? It looks like it doesn't quite fit into the housing.

Hey nik 

I am using this one 

```
Bewinner Camera Module Board for Raspberry Pi B 3/2 2592 x 1944 Resolution 5MP High Resolution, Uses OV5647 Chip, Wide Angle of 175°
```