nikivanov/watney #25: Raspberry pi 4 compatible and STL file for the 4?

lyonrt, 2022-01-22:
Morning,

Is this compatible in with the raspberry 4?  Also do you have a 3d printer file for the 4?

nikivanov, 2022-01-25:
Hi. No, the current design is only compatible with rpi 3a+. It MAY be compatible with rpi zero 2, since it's pretty much the same thing as 3a+, but I havent tested it. I dont know what you mean by "Also do you have a 3d printer file for the 4" - if you're asking whether I have a CAD model of watney that works with rpi 4, I do not, sorry! If you end up making one, feel free to create a pull request!

Syndesi, 2022-01-27:
Hi,
would it be possible to publish the CAD-files for the current design, so that changes can be made faster than with the current STLs?
I also did see that you had [published the CAD-files for the previous design](https://github.com/nikivanov/watney/commit/674829ed6d9cad4c1d8d289af444f51b6bbcd62b#diff-6ecde6f31c9523024bc5564b2cff895a5e0c3f79c34fb850e5611ca378e2da65).
Regards, Syndesi

Akzotus, 2022-01-27:
I would also appreciate the CAD files for the current one if possible. There are minor changes here and there that make working on it significantly easier when given access to the editable files rather than dealing with the STLs.
The CAD files were immeasurably beneficial when working on the previous iteration.

nikivanov, 2022-01-27:
Theres a STEP file in the root of the STLs folder - it has everything you need

Akzotus, 2022-01-27:
Well I'll be darned.

closed 2022-01-28 by nikivanov

nikivanov, 2022-01-31:
@Akzotus @Syndesi If you modify the existing version, please send me a picture of it! I'd love to see how people modify their Watneys. Please post it here: https://github.com/nikivanov/watney/issues/23