nikivanov/watney #27: Running on Raspberry Pi 4B

scifiguy000, 2022-02-25:
Great project!  I've been having fun building it (it's quite an elegant design).  Many thanks to the originator(s) for their hard work!

I did some minor mechanical modifications and used a RPI 4B.  The image boots, and WiFi can be assigned at http://192.168.4.1.  After a reboot, I see it connected to my router, and I can SSH into it at its assigned IP address.  However, the web interface does not load, and the boot up Watney voice doesn't play.  I've verified the amplifier and speaker wiring, but since the web interface does not load, I suspect something more basic is going wrong.  Any ideas what to try?

Thanks.

--Mike

nikivanov, 2022-02-28:
Hey Mike! I already responded to you on thingiverse. After Watney starts up, ssh into it, type in "sudo systemctl status watney" and paste in the output here. If there are any errors on startup, you should be able to see them there. You can also stop the service "sudo systemctl stop watney" and then run it manually from ~/watney using "python3 server.py"

nikivanov, 2022-02-28:
Both 3A+ and 4B are ARMV8 architecture, but you might need to recompile for 4B. I used [packer-builder-arm](https://github.com/mkaczanowski/packer-builder-arm) and it lists:
```
raspberry-pi-3 (Archlinux ARM (armv8))
raspberry-pi-4 (Archlinux ARM (armv7), Ubuntu 20.04 LTS))
```

I'm not sure why RPI4 is armv7 there, it's armv8 last time I checked, but either way you might need to change that and rebuild

scifiguy000, 2022-02-28:
Ah, okay.  I'll try recompiling.

I answered on Thingiverse, but to keep this thread complete, here are some results:

pi@watney:~ $ sudo systemctl status watney
● watney.service - Watney Service
   Loaded: loaded (/etc/systemd/system/watney.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2022-02-28 23:29:50 GMT; 14ms ago
 Main PID: 930 (python3)
    Tasks: 0 (limit: 3596)
   CGroup: /system.slice/watney.service
           └─930 /usr/bin/python3 -u /home/pi/watney/server.py

Feb 28 23:29:50 watney systemd[1]: Started Watney Service.

pi@watney:~ $ sudo systemctl stop watney
pi@watney:~ $ cd ~/watney
pi@watney:~/watney $ python3 server.py
Traceback (most recent call last):
  File "server.py", line 7, in <module>
    from lightscontroller import LightsController
  File "/home/pi/watney/lightscontroller.py", line 2, in <module>
    from apa102_pi.driver import apa102
  File "/usr/local/lib/python3.7/dist-packages/apa102_pi/driver/apa102.py", line 5, in <module>
    import board
  File "/usr/local/lib/python3.7/dist-packages/board.py", line 289, in <module>
    platform.system(), package[0], package[1]
NotImplementedError: Adafruit-PlatformDetect version 3.19.2 was unable to identify the board and/or microcontroller running the Linux platform. Please be sure you have the latest packages running: 'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'
pi@watney:~/watney $

nikivanov, 2022-03-01:
Yeah that looks like an issue with a platform-specific build. You really just need to reinstall apa102-pi which I'm installing here: https://github.com/nikivanov/watney/blob/master/packer/watney-image.json#L86 . My guess is that when you install it, it compiles some code that's platform-specific.

I'm curious - where did you put the amp / motor controller if you're using RPI 4? Did you make standoffs / put them on top of the usb connectors?

scifiguy000, 2022-03-01:
I used the PCB carrier you designed, but cut off the center horizontal extension, and removed the standoff underneath.  I then used 3M foam tape to attach it to the top of the USB connectors, and bent the header pins to make them right angle (I could have used right angle pins from the start, but didn't think that far ahead).

I'm learning how to use Packer Builder, and it's slow going.  I also have shoulder surgery tomorrow, so expect some delay before I have results to report.  Once I get it to run properly, I'll assemble it and post pictures.

nikivanov, 2022-03-01:
Good luck with your surgery! When you're back, check out this: https://linuxhit.com/build-a-raspberry-pi-image-packer-packer-builder-arm/ I used it as a guide for building watney's image using packer.

scifiguy000, 2022-03-08:
I'm back, mostly recovered from my surgery (biceps tenodesis, with some additional shoulder work).  Now that I can use both arms, I got Watney working with the Pi4.  You were right, nikivanov, all it took was "pip3 install -I aiohttp apa102-pi psutil pyalsaaudio smbus" from a shell, and Watney booted up and played the voice message.  I'll do the final assembly, and post pictures.  Thank you!

nikivanov, 2022-03-12:
Hey Mike,

Glad to hear your surgery went well. Thanks for confirming how to get Watney to work with RPI4. I'll close this issue and update the main README page with a link to this thread.

Nik

scifiguy000, 2022-03-15:
It appears I spoke too soon.  I got excited because the boot voice and streamer worked, but it seems that's all that is working.  Motors don't run, and headlights don't turn on.

pi@watney:~ $ sudo systemctl status watney
● watney.service - Watney Service
   Loaded: loaded (/etc/systemd/system/watney.service; enabled; vendor preset: e
   Active: active (running) since Tue 2022-03-15 22:09:38 GMT; 5min ago
 Main PID: 651 (python3)
    Tasks: 43 (limit: 3596)
   CGroup: /system.slice/watney.service
           ├─651 /usr/bin/python3 -u /home/pi/watney/server.py
           ├─660 /bin/sh -c /opt/janus/bin/janus -F /opt/janus/etc/janus/
           └─661 /opt/janus/bin/janus -F /opt/janus/etc/janus/

Mar 15 22:09:45 watney python3[651]: Saying 'Watney... Online'
Mar 15 22:09:49 watney python3[651]: New janus session 7139316904631356
Mar 15 22:09:49 watney python3[651]: Firing Session Started event
Mar 15 22:09:49 watney python3[651]: Executing '/home/pi/watney/video.sh'
Mar 15 22:09:49 watney python3[651]: Logging to '/home/pi/watney/video.log'
Mar 15 22:09:49 watney python3[651]: Starting audio pipeline
Mar 15 22:09:49 watney python3[651]: Audio pipeline started
Mar 15 22:09:50 watney python3[651]: Removed Janus session 7139316904631356
Mar 15 22:09:50 watney python3[651]: Firing Session Ended event
Mar 15 22:09:50 watney python3[651]: Stopping audio pipeline

nikivanov, 2022-03-16:
Let's try to diagnose motors first: `sudo systemctl status pigpiod` - might make sense to restart watney and do it soon after bootup, so the logs don't get rotated. The service should be up and running. Does the camera move up and down? Both the motors and the camera are driven by PWM, so if one works but the other doesnt pigpio is not the culprit.

scifiguy000, 2022-03-16:
Thanks, nikivanov.  I really appreciate your help.

The camera does move when I use the web interface to control it.

pi@watney:~ $ sudo systemctl status pigpiod
● pigpiod.service - Daemon required to control GPIO pins via pigpio
   Loaded: loaded (/lib/systemd/system/pigpiod.service; enabled; vendor preset:
   Active: active (running) since Tue 2022-03-15 22:17:20 GMT; 18h ago
  Process: 388 ExecStart=/usr/bin/pigpiod -l -t 0 (code=exited, status=0/SUCCESS
 Main PID: 405 (pigpiod)
    Tasks: 6 (limit: 3596)
   CGroup: /system.slice/pigpiod.service
           └─405 /usr/bin/pigpiod -l -t 0

Mar 15 22:17:20 watney systemd[1]: Starting Daemon required to control GPIO pins
Mar 15 22:17:20 watney systemd[1]: Started Daemon required to control GPIO pins
lines 1-11/11 (END)

nikivanov, 2022-03-17:
If the camera moves that means pigpiod can generate PWM correctly. Please check your wiring: make sure your Vin / Gnd are not reversed on the motor controller (it should have a red LED on when it's plugged in). Also check the GPIO pins, but if NO motors spin at all whether you press up or down, then it's probably a power issue. I'd also make sure you have 5V on the USB pins into which you're plugging in your motor controller. Finally, just make sure that the jumper wires aren't broken. Basically get your multimeter out and start poking :)

scifiguy000, 2022-03-17:
I had previously verified correct power and polarity to everything with a DVM, but I just discovered 2 issues with the wiring:

On a hunch, I swapped data and clock on the LED strip, and it started working.  The connections are correct to the Pi4, but it appears "D1" and "C1" on my bargain AP102 LED strip from a Tindie seller are mislabeled.

The problem with the drive motors was a bad jumper on EEP.  I swapped with another one, and it works.  I'm guessing I'd probably find a pin is crimped to insulation instead of wire.

All is working now.  Sorry for unnecessarily reopening an issue. 

Thank you!

nikivanov, 2022-03-17:
No worries, glad you got there. I'll keep it open for now - please confirm once everything is working and I'll close it

scifiguy000, 2022-03-17:
I have everything working, except the I2S microphone.  I was unable to get the brand in the BOM, unless I wanted to wait for a shipment from China, so I used the Adafruit I2S mic (https://www.adafruit.com/product/3421), and wired it accordingly.  Should the Watney mic be always on, or does it have to be enabled somehow?

nikivanov, 2022-03-18:
The mic is never enabled or disabled - it's more like whether RPi is running code to collect the data from it or not. If you open Watney's page and the video stream starts playing, you should also have the audio from the mic. Adafruit I2S mic should work - I actually used it in the previous version of Watney. The problem with it is that sometimes it produces white noise instead of normal recording. I never figured out why. I think it may have to do with the 3V line on RPi but I never confirmed it.

One thing I'd check is the channel selection on the mic. As you can see here in the video https://youtu.be/wV26r6FtXRw?t=969 LR pin of the microphone is connected to 3V and the microphone itself says "LR=H - Right" meaning if LR pin is high, then it'll use the right channel. You can also find the amp documentation (https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp/pinouts) that says "If the voltage on SD is higher than 1.4V then the output is the Left channel." The mic and the amp are both I2S devices sharing one I2S bus of the Raspberry Pi. I use the Right I2S channel for recording and Left I2S channel for playback. If you're using a different mic, make sure that that mic is also set to Right channel only. 

Looking at your mic's documentation (https://learn.adafruit.com/adafruit-i2s-mems-microphone-breakout/pinouts) it seems SEL pin also has to be set High for it to use the Right channel.

Hope that helps!

nikivanov, 2022-03-22:
Oh and another thing - you can easily test the mic by using `testrecord.sh` found in ~/watney : https://github.com/nikivanov/watney/blob/master/testrecord.sh

It'll record from the mic and save it to test.ogg in that same folder. You can use testplayback.sh to play it on Watney itself as long as the amp is working and the volume is up. Otherwise you can download it to your computer and play it there.

nikivanov, 2022-06-29:
Closing due to inactivity. @scifiguy000 let me know if you're still having trouble with it

closed 2022-06-29 by nikivanov

nicolasbeni41, 2024-03-06:
Hello my raspberry pi 3a+ does not boot with the IMG provided but with a normal raspbian distribution yes what should I do?

4 green flash + 7 green flash

start*.elf not found
Kernel image (kernel.img) not found