nikivanov/watney #28: Dock missing from Watney 2.0 STLs

rtaylo, 2022-03-11:
would you be able to post the dock STL for the 2.0 version of Watney? I have to reprint it.
(I'm also waiting on some part to build my 3.0 version, can't wait!)

nikivanov, 2022-03-12:
Oh wow, I never noticed that the dock was missing. I'll upload it in a few days and link you to it.

rtaylo, 2022-03-13:
Awesome! Thanks. I just needed to reprint it and modify it some. I'm using
a different wireless charger and need a bit more space underneath.

nikivanov, 2022-03-13:
Hey @rtaylo I've uploaded the missing dock files. Since you're going to be modifying the dock, I figured you might want the STEP file, so here are both the STL and STEP files for the dock:

https://github.com/nikivanov/watney/blob/watney2_0_dock/STLs/dock.stl
https://github.com/nikivanov/watney/blob/watney2_0_dock/STLs/dock%20v11.step

closed 2022-03-13 by nikivanov

rtaylo, 2022-03-13:
Awesome! Thanks mate. I need to post a picture as well of my Watney garage.

(Building my 3.0 soon)