nikivanov/watney #29: Watney 2.0 Constant Reboot While on Wireless Charger

rtaylo, 2022-03-18:
Hi Nikivanov,

Have you ever encountered an issue where Watney 2.0 reboots while connected to the wireless charger? 
I'm thinking about retrofitting the Watney 2.0 with the new charging contacts. 

Thanks

nikivanov, 2022-03-19:
Yes, this is one of the reasons for the redesign of Watney 3.0. The UPS used in 2.0 used to work as a true UPS in an older hardware revision. It also had 3 5V outputs and 3 3V outputs. It was also cheap to boot. It was perfect. However, the hardware was revised and the new revision does not function as a true UPS: when you supply 5V power to the UPS, the power to the devices would get cut off for about 200ms or so.

rtaylo, 2022-03-19:
I actually managed to get around the UPS issue by putting Watney on a wifi
smart plug. This allowed me to remotely turn on/off the plug. But my motors
arrived today for the Watney 3.0 build. Thanks for the input sir! Can't
wait to build the new version

nikivanov, 2022-03-20:
Good luck! Let me know how it goes.

closed 2022-03-20 by nikivanov