nikivanov/watney #30: Feature request: Additional servo control

sarreck, 2022-04-10:
Hi,

Thank you for the great project. 

It would be really useful if there was an additional preconfigured servo PIN to allow additional functionality (e.g. left/right camera, turret or RC LED control).

There would also need to be associated keyboard keys for the web UI.

In my case I want to control a RC car type LED which is controlled as a servo (works fine when connected to the camera servo in testing)

[image]

Thank you for considering this.

nikivanov, 2022-04-11:
Hi! To confirm, it sounds like you want another servo control bound to a new pair of keys. Is that correct?

nikivanov, 2022-04-11:
Do you know how to code? I can point you to a few files that you'd need to modify to make this happen.

sarreck, 2022-04-13:
> Hi! To confirm, it sounds like you want another servo control bound to a new pair of keys. Is that correct?

Yes thats correct.

sarreck, 2022-04-13:
> Do you know how to code? I can point you to a few files that you'd need to modify to make this happen.

I have tried based on your response to an older issue, however I was not able to get it to work as I didn't really know what I was doing.

nikivanov, 2022-04-13:
It shouldnt be difficult. I'll create a new branch and you should be able to copy the new files. I'll let you know when it's ready for you to test.

sarreck, 2022-04-13:
Thank you, I appreciate it.

nikivanov, 2022-04-20:
Hey @sarreck! I've made a new branch for your auxiliary servo. You can find here it: https://github.com/nikivanov/watney/tree/auxservo

By default the aux servo is on pin 6 and up/down is D / C

sarreck, 2022-04-23:
Thanks, that works great!

closed 2022-04-23 by sarreck