nikivanov/watney #32: Watney Microphone Loud Static, No sound

rtaylo, 2022-08-01:
I ordered a replacement mic module thinking that was the issue. But even after swapping the micrphone and double checking my wiring  all I'm hearing on the Watney web interface is loud static. Any suggestions?

nikivanov, 2022-08-03:
Hey @rtaylo ! I've seen this happen a few times while I was developing Watney. There are a few suspects:
- Are you using the recommended microphone - the one matching the picture from BOM? Or are you using SPH0645? I've had nothing but issues with the latter, which is why I switched to the current one.
- If you are using the mic from BOM, make sure the wiring is correct when it comes to L/R selection - the microphone LR pin should be connected to a 3.3V pin on Raspberry pi, which forces it to use the left channel. At the same time, the amp's SD pin should be connected according to [the diagram](https://github.com/nikivanov/watney/blob/master/images/wiring/audio%20wiring.png) - this will force the amp to receive on the right channel. If either get crossed up, you'll get static
- One of my microphone boards simply did not produce anything but static. I chucked it, since doing a drop in replacement with another board solved the issue
- If nothing else works, do a test in isolation - install plain raspbian on your raspberry pi, hook up just the microphone and try recording

rtaylo, 2022-08-04:
Here is the microphone I purchased.
I printed out the wiring diagram and followed it closely. Does this seem
like the right microphone?

https://m.aliexpress.com/item/2251832824703750.html?spm=a2g0n.order_detail.0.0.32cef19cur1lzU

nikivanov, 2022-08-04:
That looks to be the right microphone. I suggest you install vanilla raspbian on your RPI and make sure the mic works OK in that environment.

rtaylo, 2022-08-09:
Good thinking, I will do that and report back

nurglexp, 2022-10-15:
I have exact same issue with https://www.aliexpress.com/item/4001244371729.html

Do you have a link to a known working microphone?

rtaylo, 2022-10-15:
I still haven't been able to get this working

nikivanov, 2022-10-20:
Hi @nurglexp and @rtaylo. @nurglexp this is the correct microphone. Generally, when I buy parts from aliexpress I buy a bunch, 3-5 depending on the price. Usually they are cheap enough that I'd rather pay a bit more money than order one, wait a month, get a part that's potentially broken and then have to order again. I ordered 5 of these microphones and had this exact issue with one of them. To be fair, I had this same issue but more intermittent with the Adafruit I2S microphone that I used in the previous version of Watney. However, with these mics my experience has been that they either always work or never.

My advice is to buy 1-2 more of the microphones, preferably from a different supplier, and see if you can resolve the issue by simply replacing the part.

Another thing you can try is to disconnect everything but the microphone from the raspberry pi, leave it unpowered for a few hours (more on that later), start it up, SSH into watney and run [this script](https://github.com/nikivanov/watney/blob/master/testrecord.sh) It will record from the mic into `test.ogg` file. Transfer it to your PC and see if it's noise, silence, or actual sound recording.

Let me know how that goes for you!

rtaylo, 2022-10-20:
Will do, thanks Nik! I've actually got a new baby and I'm about to start
heading back into the office. I'll be using Watney to spend time with the
wife and baby while I'm at work lol

nikivanov, 2022-10-23:
@rtaylo congrats! Coincidentally, I just got a brand new baby as well! I'm hoping to get some time to work on a new project of mine during my paternity leave (stay tuned), though so far it's been pretty baby-centric, as expected.

rtaylo, 2022-10-24:
Congratulations to you as well!! Our little one just hit one month old
yesterday. Sleep is getting pretty hectic lol. Looking forward to see what
new projects you come up with!

rtaylo, 2022-10-26:
@nurglexp I was looking on Ali Express and someone seems to be selling a "new" version of the Sipeed microphone.
https://bit.ly/3FfLEH6
Ordered a few to test out, I'll report back

rtaylo, 2022-10-26:
Whoops sorry about that.
Try this:
https://www.aliexpress.us/item/3256804659411651.html?spm=a2g0o.productlist.0.0.63b812593zLp7P&algo_pvid=233c5f61-5917-451c-ae98-dfc271e8fd38&algo_exp_id=233c5f61-5917-451c-ae98-dfc271e8fd38-1&pdp_ext_f=%7B%22sku_id%22%3A%2212000030719724355%22%7D&pdp_npi=2%40dis%21USD%212.78%211.81%21%21%21%21%21%40210318b916668204406674262e4389%2112000030719724355%21sea&curPageLogUid=QW33FW1Rzlte

nikivanov, 2022-10-26:
That looks like a drop in replacement. Please let me know how it goes and I
can update the BOM if all goes well. Thanks!

nurglexp, 2022-12-03:
@rtaylo any luck with the replacement microphone?

nikivanov, 2022-12-14:
I'm also very curious about this new version!

rtaylo, 2023-02-04:
@nurglexp the new microphone fixed the issue!

closed 2023-02-04 by rtaylo

nurglexp, 2023-02-04:
Amazing, thanks for the confirmation. Ordered the replacement microphone 👯

nikivanov, 2023-02-04:
I'd recommend ordering a couple, especially if ordering from AliExpress -
quality control seems to be pretty poor with them. Glad this worked out!