nikivanov/watney #33: Issues with UPS communicating with Raspberry PI

EGimagination, 2023-04-28:
Hello, I'm having a single issue with Watneys UPS comms. In order to get it communicating/ running I had to change the address from 0x17 to 0x68 and the software works with one issue. 
[image]
The battery reads 42% no matter the actual battery voltage is. It also does not sense when its charging making it hard to tell weather or not its going to survive the night haha.
[image]

I assume its something to do with the raspberry pi not communicating with the UPS? all the parts are identical to the BOM. Any help would be greatly appreciated! 

Thanks, 
Eric

nikivanov, 2023-04-29:
Hi Eric!

First of all, just wanted to say thank you for making a Watney for kids to learn 3D printing and get into STEM - that is truly heartwarming.

In terms of changing the address, I don't think this is correct. Per documentation, the address is indeed 0x17: https://wiki.52pi.com/index.php/EP-0136#USB_Plus_V5.0_Register_Mapping_Chart . 0x68 looks to be the RTC module, so you not getting the correct data is to be expected.

What error do you get when the address is 0x17? Are you absolutely sure you bought UPS-Plus (EP-0136) and not UPS (EP-0118) ?

nikivanov, 2023-04-29:
Additionally, see what results you get when you scan for available i2c devices: https://learn.adafruit.com/scanning-i2c-addresses/raspberry-pi

nerdymark, 2023-04-30:
My 0x17 is no longer visible. I think the board went bad, other addresses are visible. Maybe we're having the same problem. I'm looking for an alternative UPS solution because this will be my 2nd one to have failed.

EGimagination, 2023-04-30:
Thanks for the quick reply! Good catch on the model numbers the one I have is the EP-0118. I'm getting better results. If you have any advice on how to properly read the values with the powerplant.py portion of the code would be greatly appreciated!

This is the result from the I2C scan
[image]

I used some demo code to confirm that the device was communicating properly.
[image]

Then updated the powerplant.py address to 0x40 & this is what the battery portion is showing me.
[image]

nikivanov, 2023-05-01:
I'm afraid EP-0118 is simply not compatible with the features you're looking to enable: it only reads the voltage on the output of the 5V regulator, which is, unsurprisingly, 5V regardless of the state of charge of the batteries or the input voltage from the charger. Therefore, you won't be able to see the battery charge or the charging state unless you upgrade to EP-0136

nikivanov, 2023-05-01:
@nerdymark that's strange that they keep breaking on you. Did you flash the v10 firmware on your EP-0136?

nerdymark, 2023-05-02:
> @nerdymark that's strange that they keep breaking on you. Did you flash the v10 firmware on your EP-0136?

The first one I broke by doing something I shouldn't have done: un-mount it while batteries were installed. I bridged a connection and let out some magic smoke.
Second one I'm sure I broke too. It turns out my batteries were very dead, so I put different ones in it and ran the self-diag on it. The charger is not charging the batteries. I think I broke it by having the pi and the board both plugged into a USB-C power supply at the same time. 
Third one is arriving today, lessons learned :)

nerdymark, 2023-05-02:
[image]
Mine's working again with the replacement upsplus. 
ServoCity Runt Rover platform, a couple of hats with the motor controllers, speaker amps, etc

nikivanov, 2023-05-06:
@nerdymark so you're running the software on a different hardware chassis? Awesome! How's tank turning working with 3 wheels per side? Does your camera tilt up and down?

nerdymark, 2023-05-06:
> @nerdymark so you're running the software on a different hardware chassis? Awesome! How's tank turning working with 3 wheels per side? Does your camera tilt up and down?

It turns fine though I'm sure it could be better. The wheels are powered by the big battery in the back that can output 5 - 12v. I have 3 motor controllers sharing 9 volts from it. The camera tilts up and down.

The echo sensor is experimental, I had modified the motor controller to not move the rover if it's too close to something.