nikivanov/watney #34: Raspberry Pi Camera 3 incompatibility

camrichmond, 2023-07-21:
I recently started building a Watney, and I've run into an issue with the new official Raspberry Pi camera module V3. Unfortunately these new modules aren't compatible with Buster and raspivid - instead they require the new drivers from Bullseye and libcamera or Picamera2.

I've made some progress by upgrading the installed Watney image to Buster. I can now see my camera module with a vcgencmd get_camera, but there are some broken dependencies when trying to install libcamera-apps that I still need to resolve. Someone more savvy than me could probably make this change easily, but I might just make it easier on myself and buy an older wide-angle camera module. FWIW my old 1st generation camera module works perfectly right out of the box with the stock Watney image.

I just wanted to let anyone who might be running into the same issue know that the newer V3 cameras won't work with the current Watney image. Keep up the great work - this is a really fun project!

camrichmond, 2023-07-21:
Update - after resolving some dependencies and modifying the video.sh file to use libcamera-vid instead of raspivid, it works! Because libcamera was built to eschew the proprietary Broadcom GPU-reliant legacy apps, most of the options remain the same, but here are the updated contents of video.sh that work on Bullseye:

`libcamera-vid -n --width 1280 --height 720 --bitrate 2500000 --framerate 25 -t 0 --profile baseline --intra 25 --flush -o -| gst-launch-1.0 fdsrc do-timestamp=true ! h264parse ! rtph264pay config-interval=1 pt=96 ! udpsink host=127.0.0.1 port=8004`

Hopefully that's helpful to someone!

nikivanov, 2023-07-27:
Thanks @camrichmond! I'll make a note on the main page regarding this. How is the new camera module working for you? I found the non-fisheye cameras a bit too tough to drive with, since you're not seeing the environment right next to the rover.

closed 2023-08-01 by nikivanov