nikivanov/watney #36: Use Docker

eerison, 2023-09-24:
Hi @nikivanov 

I wa wondering... have you considered to use docker instead of packer? Why?

-  Because packer could be used just for few boards,
- And when you release a new image I must reinstall the SO,  and with docker I just need to pull the new image,  and In case there is some issue, I could revert easily.

nikivanov, 2023-09-25:
Hey @eerison!

The code uses pigpio which only works with raspberry pis from what I know, so making Watney more board-agnostic would involve more than just a docker image. Beyond that, I simply didn't see any advantage to using docker - it may not require extra compute, but it certainly wouldn't make the code more performant than running the process natively.

That being said, feel free to try it yourself and if it works, please open a PR and I can provide the docker image in addition to the native SD card image.

eerison, 2023-09-25:
I see,  I am not saying that docker gonna be more performing,  but the current image just works for raspberry...

Just to know,  do you have any suggestion , to make the project works for more boards? Like banana pi or orange pi

eerison, 2023-09-25:
> I see, I am not saying that docker gonna be more performing, but the current image just works for raspberry...
> 
> Just to know, do you have any suggestion , to make the project works for more boards? Like banana pi or orange pigpio works only for raspberry? 😶‍🌫️

And why pi

eerison, 2023-09-25:
An regarding for pigpio , is it used just to integrate with pins, or is it used somewhere else?

Is it used in one place,  or many places in the code?

And do you think pigpio is the biggest impediment?

I was thinking to use armbian, maybe it could be an alternative to support different boards

nikivanov, 2023-09-25:
PIGPIO, raspivid, googlevoicehat-soundcard are all raspberry pi native, to name a few.

The board would also have to support hardware video encoding, I2C, SPI and I2S. OrangePi might support all that but it'd be a good amount of work to make Watney board-agnostic

closed 2023-10-04 by nikivanov