nikivanov/watney #37: Change wifi password

eerison, 2023-10-04:
Hey @nikivanov 

after authenticate in the wifi, is it possible to change the password again?

eerison, 2023-10-04:
other question do you have any idea why it does not work access via proxy?

I am adding it in traffic but it is not working :/, any idea why? does it work with other host or just watney.local?

nikivanov, 2023-10-04:
If the password of the access point changes, Watney will no longer be able to connect to it. Whenever Watney boots up and cannot connect to any of its known networks, it restarts in AP mode as mentioned [here](https://github.com/nikivanov/watney#configuration)

As far as proxy access goes, watney.local may not work because mDNS resolution may not work through the proxy. You need to find Watney's IP as it is seen my your client computer that's connected to the proxy and use that IP instead of watney.local .  I use [Zerotier](https://github.com/nikivanov/watney#remote-access) for remote access of my Watneys and mDNS works just fine with it

eerison, 2023-10-04:
Ok cool, via ip it worked, I am just facing a issue related with janus, I guess I just need to add this port also

POST https://sub.mydomain.com:8089/janus net::ERR_CONNECTION_REFUSED

is there any way to access like 

https://sub-janus.mydomain.com/janus instead of https://sub.mydomain.com:8089/janus ?

because I do not expose ports just 443

nikivanov, 2023-10-04:
That would be a question for Janus. I use zerotier because it takes care of
all the ports. I'm not sure how to set up Janus via a proxy.

eerison, 2023-10-05:
When you open watney via your proxy 

Do you have a request for 8089/janus? If yes then you expose 8089 port right?

eerison, 2023-10-05:
Ok I found where it should be changed 

https://github.com/nikivanov/watney/blob/master/js/watney_janus.js#L50

Is there anyway to get config variables in this file?

nikivanov, 2023-10-05:
I'm not sure what you're asking. You can modify this file as you see fit
locally. What config variables are you trying to get and in what context?

eerison, 2023-10-06:
> I'm not sure what you're asking. You can modify this file as you see fit locally. What config variables are you trying to get and in what context?
> […](#)

Ok I know that I could access the raspberry and change this file directly, and solve my issue. 

But don't you think that others people gonna face the same issue later?

My idea is add [.env](https://www.npmjs.com/package/dotenv) in the project and get this host from there like 

SERVER_HOST_JANUS=https://my.domain.com

If this env is set, then I use this host,  instead of the current one. 

Does it make sense?

eerison, 2023-10-06:
> Ok I found where it should be changed
> 
> https://github.com/nikivanov/watney/blob/master/js/watney_janus.js#L50
> 
> Is there anyway to get config variables in this file?

I changed this line into the server, but the client keep doing request to the same host, the changes did not apply.
any idea why? is there any cache that I need to clean?

nikivanov, 2023-10-06:
No idea. With respect, you're not encountering an issue with Watney as it was designed to be used. You're trying to achieve an unsupported use case in this post and your other posts regarding other ARM boards, packer vs docker etc. You're welcome to try to figure it out yourself and if you get it to work, I'd be happy to merge it in, but I'm afraid I don't have time to help you add these new use cases.

closed 2023-10-06 by nikivanov

eerison, 2023-10-06:
But I'm Testing with a raspberry pi 3b+ 😕

nikivanov, 2023-10-06:
Only raspberry pi 3a+ is supported, though my guess is that 3b+ should work as well. But this is a great example: if 3b+ ends up not working for you and you'd like to make it work, this wouldn't be an issue with Watney, as it's not designed for it. You're welcome to try to make it work with 3b+, but it would be on you to figure out how. 

In general, I don't want to discourage you from trying to make Watney work outside of its supported use cases, but you keep opening issues with generic "how do I do X?" questions and the answer to all of them is "It's not supported. If you'd like it to be, you need to figure it out". I hope I'm not sounding rude - that's not the intention. It's just I've made all the improvements I wanted to this project and don't want to spend more time on it beyond supporting existing documented use cases.

eerison, 2023-10-28:
Hey @nikivanov 

do you have any idea how this watney.local is configurated.
when I connect into my local network (via vpn), I can ping local ip, watney.lan, but I can not ping at watney.local.

maybe if I found this issue, maybe  I could find the solution why It is working just for audio/speaker, and not for video.

nikivanov, 2023-10-28:
watney.local is just mDNS. The raspberry pi image sets the hostname of the
device to watney which makes it accessible at watney.local