nikivanov/watney #40: name 'AudioManager' is not defined

eerison, 2023-12-15:
Did you face this issue? note I am using the default image

```
Dec 15 22:20:03 watney python3[807]: Using TLS with keys in '/home/pi'
Dec 15 22:20:03 watney python3[807]: Traceback (most recent call last):
Dec 15 22:20:03 watney python3[807]:   File "/home/pi/watney/server.py", line 173, in <module>
Dec 15 22:20:03 watney python3[807]:     audioManager = AudioManager(config)
Dec 15 22:20:03 watney python3[807]: NameError: name 'AudioManager' is not defined
Dec 15 22:20:03 watney systemd[1]: watney.service: Main process exited, code=exited, status=1/FAILURE
Dec 15 22:20:03 watney systemd[1]: watney.service: Failed with result 'exit-code'.
Dec 15 22:20:04 watney systemd[1]: watney.service: Service RestartSec=1s expired, scheduling restart.
Dec 15 22:20:04 watney systemd[1]: watney.service: Scheduled restart job, restart counter is at 21.
Dec 15 22:20:04 watney systemd[1]: Stopped Watney Service.
Dec 15 22:20:04 watney systemd[1]: Started Watney Service.
```

nikivanov, 2023-12-18:
I'm not sure how you can be using the default Watney image and have an import error. Please go into `watney` folder, do `git status` and post the output

eerison, 2023-12-18:
Now I am getting this error:

```
Dec 18 20:57:36 watney systemd[1]: Started Watney Service.
Dec 18 20:57:38 watney python3[1358]: Using TLS with keys in '/home/pi'
Dec 18 20:57:38 watney python3[1358]: Use hardware SPI
Dec 18 20:57:38 watney python3[1358]: Traceback (most recent call last):
Dec 18 20:57:38 watney python3[1358]:   File "/home/pi/watney/server.py", line 184, in <module>
Dec 18 20:57:38 watney python3[1358]:     powerPlant = PowerPlant(config)
Dec 18 20:57:38 watney python3[1358]:   File "/home/pi/watney/powerplant.py", line 10, in __init__
Dec 18 20:57:38 watney python3[1358]:     self.bus.write_byte_data(DEVICE_ADDR, 17, cutoffVoltage & 0xFF)
Dec 18 20:57:38 watney python3[1358]: OSError: [Errno 121] Remote I/O error
Dec 18 20:57:38 watney systemd[1]: watney.service: Main process exited, code=exited, status=1/FAILURE
Dec 18 20:57:38 watney systemd[1]: watney.service: Failed with result 'exit-code'.
Dec 18 20:57:39 watney systemd[1]: watney.service: Service RestartSec=1s expired, scheduling restart.
Dec 18 20:57:39 watney systemd[1]: watney.service: Scheduled restart job, restart counter is at 84.
Dec 18 20:57:39 watney systemd[1]: Stopped Watney Service.
Dec 18 20:57:39 watney systemd[1]: Started Watney Service.
Dec 18 20:57:40 watney python3[1364]: Using TLS with keys in '/home/pi'
Dec 18 20:57:41 watney python3[1364]: Use hardware SPI
Dec 18 20:57:41 watney python3[1364]: Traceback (most recent call last):
Dec 18 20:57:41 watney python3[1364]:   File "/home/pi/watney/server.py", line 184, in <module>
Dec 18 20:57:41 watney python3[1364]:     powerPlant = PowerPlant(config)
Dec 18 20:57:41 watney python3[1364]:   File "/home/pi/watney/powerplant.py", line 10, in __init__
Dec 18 20:57:41 watney python3[1364]:     self.bus.write_byte_data(DEVICE_ADDR, 17, cutoffVoltage & 0xFF)
Dec 18 20:57:41 watney python3[1364]: OSError: [Errno 121] Remote I/O error
Dec 18 20:57:41 watney systemd[1]: watney.service: Main process exited, code=exited, status=1/FAILURE
Dec 18 20:57:41 watney systemd[1]: watney.service: Failed with result 'exit-code'.
Dec 18 20:57:42 watney systemd[1]: watney.service: Service RestartSec=1s expired, scheduling restart.
Dec 18 20:57:42 watney systemd[1]: watney.service: Scheduled restart job, restart counter is at 85.
Dec 18 20:57:42 watney systemd[1]: Stopped Watney Service.
Dec 18 20:57:42 watney systemd[1]: Started Watney Service.
Dec 18 20:57:43 watney python3[1371]: Using TLS with keys in '/home/pi'
Dec 18 20:57:43 watney python3[1371]: Use hardware SPI
Dec 18 20:57:43 watney python3[1371]: Traceback (most recent call last):
Dec 18 20:57:43 watney python3[1371]:   File "/home/pi/watney/server.py", line 184, in <module>
Dec 18 20:57:43 watney python3[1371]:     powerPlant = PowerPlant(config)
Dec 18 20:57:43 watney python3[1371]:   File "/home/pi/watney/powerplant.py", line 10, in __init__
Dec 18 20:57:43 watney python3[1371]:     self.bus.write_byte_data(DEVICE_ADDR, 17, cutoffVoltage & 0xFF)
Dec 18 20:57:43 watney python3[1371]: OSError: [Errno 121] Remote I/O error
Dec 18 20:57:43 watney systemd[1]: watney.service: Main process exited, code=exited, status=1/FAILURE
Dec 18 20:57:43 watney systemd[1]: watney.service: Failed with result 'exit-code'.
Dec 18 20:57:45 watney systemd[1]: watney.service: Service RestartSec=1s expired, scheduling restart.
Dec 18 20:57:45 watney systemd[1]: watney.service: Scheduled restart job, restart counter is at 86.
Dec 18 20:57:45 watney systemd[1]: Stopped Watney Service.
```

```
pi@watney:~/watney $ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
pi@watney:~/watney $ ^C
pi@watney:~/watney $ git remote -v
origin	https://github.com/nikivanov/watney.git (fetch)
origin	https://github.com/nikivanov/watney.git (push)
pi@watney:~/watney $ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
pi@watney:~/watney $ git branch
* master
```

Context:

Why it was working and now not anymore?

the battery uncharged completely, and when I charged again it was turn on and off all the time, then I realised that the issue was: raspberry and all things was trying to turn on and couldn't because it hasn't enough battery.

then I disconnected the raspberry and then I could charge the battery to 100%, but after this it doesn't work anymore :/.

Note: power button is not working too :(, maybe I should reset the UPS? if yes how 👀 ?

eerison, 2023-12-22:
Watney is alive again, thank you <3

closed 2023-12-22 by eerison