nikivanov/watney #41: Audio issue

eerison, 2023-12-22:
After upgrade UPS fw, audio is not working anymore :/

sometime I hear "Watney online", but after few seconds I hear like audio cut, and it stops to work.

I can see this log

```
Dec 22 10:53:23 watney kernel: [  163.981478] voicehat-codec voicehat-codec: Disabling audio amp...
Dec 22 10:53:23 watney kernel: [  163.981637] voicehat-codec voicehat-codec: Enabling audio amp...
Dec 22 10:53:34 watney kernel: [  174.283044] voicehat-codec voicehat-codec: Disabling audio amp...
Dec 22 10:53:34 watney kernel: [  174.283471] voicehat-codec voicehat-codec: Enabling audio amp...
Dec 22 10:53:51 watney kernel: [  191.484460] voicehat-codec voicehat-codec: Disabling audio amp...
Dec 22 10:53:51 watney kernel: [  191.485285] voicehat-codec voicehat-codec: Enabling audio amp...
Dec 22 10:54:39 watney kernel: [  240.046761] voicehat-codec voicehat-codec: Disabling audio amp...
Dec 22 10:54:39 watney kernel: [  240.052220] voicehat-codec voicehat-codec: Enabling audio amp...
Dec 22 10:54:46 watney kernel: [  247.003943] voicehat-codec voicehat-codec: Disabling audio amp...
Dec 22 10:54:46 watney kernel: [  247.007404] voicehat-codec voicehat-codec: Enabling audio amp...
```

eerison, 2023-12-22:
when I execute `./testplayback.sh`

```
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstFileSrc:filesrc0: Resource not found.
Additional debug info:
gstfilesrc.c(533): gst_file_src_start (): /GstPipeline:pipeline0/GstFileSrc:filesrc0:
No such file "test.ogg"
Setting pipeline to NULL ...
Freeing pipeline ...
```

nikivanov, 2023-12-28:
you should run testrecord first - it will record test.ogg that testplayback complains about not having

nikivanov, 2024-03-11:
Closing due to inactivity

closed 2024-03-11 by nikivanov