nikivanov/watney #42: Raspberry don't BOOT

nicolasbeni41, 2024-03-06:
Hello mon raspberry pi 3a+ ne boot avec l'IMG fourni mais avec une distribution raspbian normale oui que faire ?

bennyp3333, 2024-04-13:
The same thing happened to me. Return/RMA your raspberry pi if possible otherwise replace the start*.elf, fixup*.dat and bootcode.bin on the sd card after flashing with the respective files from [rpi-firmware-master](https://github.com/raspberrypi/rpi-firmware). At least that worked for me, YMMV.
https://www.thingiverse.com/thing:6493508

ZehnderUMN, 2024-04-20:
My pi 3a+ was not booting either. Replacing the files worked

moaazsidat, 2024-07-15:
Ran into the same with pi 3a+, replacing the files hasn't worked

nikivanov, 2024-08-01:
I'm hoping to do a 2024 model refresh of Watney with a better power plant and latest linux, but I don't have a timeline for it. Latest linux particularly should solve this issue.

ZehnderUMN, 2024-08-01:
Ok thank you, I am looking forward to it!Alan Zehnder

neuro717, 2025-03-12:
I have exact same issue, image is not working.
Replaced files, obviously something else -is- happening, but won't connect to network.

Would be great to have a refresh on Watney. There's no other telepresence rover that's so accessible, easily affordable, and cool looking =)

 and here's what's been done to identify the boot issue:

1.	Tried the ‘usual’ way after wiring, by sticking in the Balena / Pi imager created 64gb SD card, no sound (following instructions in the video)
2.	Thought it might just be the wiring/batteries, plugged the whole Raspi 3A+ out, plugged into a regular power supply with GPIO pinned to speaker, no sound
3.	Speculated that it’s the GPIO the core cause, unplugged those, no sound. Decided to measure the voltage in, its working (0.9V/0.1-0.2A) consumption on boot (as expected)
4.	Thought the Raspi might not be working, used the same SD card, flashed Raspbian with the same macOS device using Balena – all working
5.	Everything else confirmed to be working, decided to plug it into the HDMI with keyboard – no graphics
6.	Decided to force HDMI output – no image on HDMI either, so I can’t troubleshoot what’s going on
7.	Later, tried if I can ‘rebuild’ the image on a raspi lite image 64 – and gave up after a while.

nikivanov, 2025-03-12:
I appreciate the kind words! I'm very close to releasing my new project, and my plan is to do a Watney refresh for bookworm after it. I'll ping this thread once I have a new version working.

neuro717, 2025-03-28:
@nikivanov : If it takes too much time and effort to build a complete image, why not give instructions so users can add all dependencies, install apps, and do our own configs to run it? I think many on the thread can follow that.

neuro717, 2025-03-28:
Alright, gave up. Lightyears behind what's needed to get this up and running. Will wait for your next release.

nikivanov, 2025-04-02:
Hi! I know a few of you are waiting for the new image of Watney. Here's why it's taking longer than I thought: I built Watney on buster, which is really old at this point. The new debian version, bookworm, has some changes to network setup, so I can't even SSH into Watney over wifi. I need to build a new image, but a lot of setup things need to be fixed. 

It's not all bad though, there's this: https://github.com/RPi-Distro/pi-gen - this is a much more reputable way of building Pi OS images with customizations. So I need to port the customizations from the current build system to pi-gen so at least I can SSH into the OS.

neuro717, 2025-04-02:
Well, if there’s any granular work I can do to help, let me know.

When I tried building the watney image from zero (and didn’t get it all up and running, got stuck at getting the UI work (probably flask?) and control the motors), that took me a day and tons of claude help.

Appreciate what you’re putting into this.